exo-lang/exo #92: Make unification (optionally) memory aware?

alexreinking, 2021-11-14:
Should unification be memory-aware? It mixes up loads and stores when doing `replace_all`.

jrk, 2021-11-14:
Specifically, it seems like it would be less surprising if `replace_all` **defaulted** to respecting memory types, and optionally disabled that with a flag something like `ignore_memories=True`.

jrk, 2021-11-14:
Also, it may be worth noting in documentation why this exists: it is fundamentally necessary to pass through intermediate states which are invalid wrt. memory assignment constraints (but equivalent otherwise).

gilbo, 2021-11-15:
Possible method of addressing this: allow for memories to optionally matter for the front-end pattern matching, but not for unification itself.

— Gilbert