exo-lang/exo #94: Start using Python type annotations and checking

jrk, 2021-11-15:
Partly mirrors #12, but we don't necessarily have to use `dataclass`es.

For example, many of the schedule passes have a bunch of boilerplate to check the types of the arguments and raise standard error messages if they're wrong. This feels like it should be managed with type annotations and a standard type checking system.

Static type checking (mypy or similar) is also worth considering.

alexreinking, 2021-11-15:
> Static type checking (mypy or similar) is also worth considering.

I can't wait to see what `adt` does to mypy 😉

jrk, 2021-11-15:
…that's why the original issue #12 specifically combined "rewrite to use dataclasses and types" :)

I think the idea would be to have ADT generation be a static code generation process that spits out Python source, and that source would be simpler, more idiomatic dataclass declarations as much as possible.

alexreinking, 2021-11-15:
> I think the idea would be to have ADT generation be a static code generation process that spits out Python source, and that source would be simpler, more idiomatic dataclass declarations as much as possible.

That has some pros/cons... `import` statements are a hassle with generated Python sources. You need an external build system (AFAIK) to coordinate this. I anticipate better would be to use PEP 561 stub files (`.pyi`) and continue using runtime-generated classes.

jrk, 2021-11-15:
Also reasonable.

gilbo, 2021-11-15:
A tangential issue worth considering is whether ADT errors vs type errors at the exposed API should raise different kinds of exceptions so that users can distinguish a “my problem” from a “your problem”. Current convention (possibly violated?) is that internal errors should fail assertions.

rachitnigam, 2023-09-20:
We're going tag this into the bigger Exo core rewrite issue

closed 2023-09-20 by rachitnigam