exo-lang/exo #102: The simplifier is slow

alexreinking, 2021-11-17:
We need to be able to propagate "scoped facts" in the IR, of the following form:

```
if <expr> == <term>:
    <body>
```

Clearly, `<expr>` should be replaced by `<term>` in `<body>`.

The current implementation of this maintains a dictionary of `str(expr)` -> `<term>` mappings, but going through the pretty printer is _incredibly_ slow. One resolution would be to make sure that ADT / IR objects (expressions at least, if not full statements) incrementally compute and store structurally consistent hashes.

alexreinking, 2021-11-17:
One prerequisite for this might be to replace mutable data structures in the IR nodes (e.g. lists) with immutable equivalents (e.g. tuples).

gilbo, 2021-11-17:
Looks like it’s not so simple after all! =p

rachitnigam, 2023-09-20:
We will create a tracker for things that need to be optimized within the internal infrastructure.