exo-lang/exo #117: delete_config and assert_if should be part of simplify

yamaguchi1024, 2021-12-20:


skeqiqevian, 2023-09-20:
Ideally, we'd like to be able to reason about the following program: 
```
x: f32 @ DRAM
for i in seq(0, 8):
    if i + 3 > -1:
        ...
    else:
        ...
->
x: f32 @ DRAM
for i in seq(0, 8):
    ...
```

rachitnigam, 2023-09-20:
Thanks for the example! This is a classic example of range analysis which compilers can use to eliminate always true or always false branches. I think for Exo to prove this transformation correct, it would need to propagate that information forward and then conclude that the `if` is always true.

rachitnigam, 2023-09-20:
This also seems like a transformation that is always profitable and applied when the program reaches this state, that is, there is no obvious reason to want to keep the always true branch.