exo-lang/exo #155: New `stage_mem` directive will have an unhandled failure edge case

gilbo, 2022-02-03:
When the size of a buffer depends on global, mutable state, then this directive can fail because it syntactically copies the size expressions without regard to any possible mutation of global values.

One easy solution to this problem is to implement some kind of check that lets us confirm that the size expressions for a buffer do not change in value at any point during the execution of some block.  Conceivably this might inhibit some intended uses of the directive.

One solution that will likely be important in the long run is having local (non-mutable?) variable bindings for control values.  If such a feature is added, then the prior constraint becomes irrelevant.  If a user hits the above error because the size expression changes value, then they can cache the size in a local control variable, whose value will not change.

gilbo, 2022-02-05:
One idea for a conservative fix to this issue (which probably recurs in other scheduling transformations) is to simply disallow various sizing or indexing expressions from containing global/config values.  Currently, I don't know of cases where those expressions need to depend on globals/config.

Furthermore, if initial programs are always written to avoid the use of globals in their original formulation, then this kind of thing can usually be worked around

yamaguchi1024, 2022-02-16:
it'll be nice if we could name the iteration variables of initialization and writeback loops (in addition to naming them `i0`, `i1`... by default)