exo-lang/exo #172: Issue a warning if an argument in a proc is not used

yamaguchi1024, 2022-03-07:
Abhijit was saying that exo was previously complaining if an argument is not used in the proc but it doesn't anymore so he was confused. Maybe this is because of the new effect check? If I recall correctly, we never complained even the argument is not used. In both cases, we should make the behavior explicit.

adavare, 2022-03-08:
I can't confirm that it previously complained, but it certainly no longer does. I suggested that for an unused argument, if a default value is provided, then that should be used by `replace()`. If no default is provided, then any satisfying value can be inserted by the `replace()`, but a warning should be given to alert the user to a potential bug.

yamaguchi1024, 2022-03-08:
We currently don't have a default argument syntax, but that might be one option