exo-lang/exo #200: Generated C should use export macros

alexreinking, 2022-06-14:
The generated C code should stamp exported functions with `<LibName>_API` macros to explicitly mark their symbols as exported. This varies by compiler and should be user-overridable. This will be necessary to play nice on Windows at _all_ and to support Unix-y users compiling with `-fvisibility=hidden`.

gilbo, 2022-06-14:
Does this mean e.g. if I write a library that exports functions matmul2d() and conv2d(), that there would be some kind of option which would change those to matmul2d_MYLIB_API() and conv2d_MYLIB_API()?  Or do you mean something else here?

rachitnigam, 2023-09-20:
@SamirDroubi mentioned that this is probably for building dynamically compiled C libraries from Exo code. This is quite a low priority so not sure when we'd work on it.

alexreinking, 2023-09-20:
No, this is for building shared libraries correctly. This will prevent symbol resolution issues that arise when loading two shared libraries, both of which are written in Exo.