exo-lang/exo #286: reorder_stmts block is buggy

yamaguchi1024, 2022-11-11:
```
`--> exocc -o blur --stem blur blur.py
This is original blur
def blur(g: R[100] @ DRAM):
    f: R[101] @ DRAM
    for i in seq(0, 101):
        f[i] = 1.0
    for i in seq(0, 100):
        g[i] = f[i] + f[i + 1]

def blur(g: R[100] @ DRAM):
    f: R[101] @ DRAM
    for io in seq(0, 10):
        for ii in seq(0, 10):
            f[ii + 10 * io] = 1.0
    f[100] = 1.0
    for io in seq(0, 10):
        for ii in seq(0, 10):
            g[ii + 10 * io] = f[ii + 10 * io] + f[1 + ii + 10 * io]

Traceback (most recent call last):
  File "/home/yuka/.local/bin/exocc", line 8, in <module>
    sys.exit(main())
  File "/home/yuka/.local/lib/python3.9/site-packages/exo/main.py", line 37, in main
    library = [
  File "/home/yuka/.local/lib/python3.9/site-packages/exo/main.py", line 40, in <listcomp>
    for proc in get_procs_from_module(load_user_code(mod))
  File "/home/yuka/.local/lib/python3.9/site-packages/exo/main.py", line 69, in load_user_code
    loader.exec_module(user_module)
  File "<frozen importlib._bootstrap_external>", line 855, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/yuka/exo/blur.py", line 25, in <module>
    blur = reorder_stmts(blur, 'for io in _:_ #0 ; f[_] = _')
  File "/home/yuka/.local/lib/python3.9/site-packages/exo/API_scheduling.py", line 98, in __call__
    bargs[nm] = argp(bargs[nm], bargs)
  File "/home/yuka/.local/lib/python3.9/site-packages/exo/API_scheduling.py", line 436, in __call__
    self.err(
  File "/home/yuka/.local/lib/python3.9/site-packages/exo/API_scheduling.py", line 57, in err
    raise Error(
ValueError: argument 1, 'block_cursor' to reorder_stmts: expected a block of size 2, but got a block of size 1
```

SamirDroubi, 2023-01-11:
The implementation is not buggy. The pattern passed would be recognized as a for loop with a body first containing a hole and then the statement f[_]. This was recognized as a one-statement block above. To convey through the pattern that the loop's body has ended, you must add a new line `\n` character after the body `bar = reorder_stmts(bar, "for io in _:_ ;\nf[_] = _")`. Alternatively, you could use a multiline string.

SamirDroubi, 2023-01-11:
I made a PR with a simplified version of the test above to have an example of this in the codebase and to also add a test for `reorder_stmts` since there wasn't one in `test_schedules.py`

closed 2023-01-13 by SamirDroubi