exo-lang/exo #319: Implement simplify with SMT solver

yamaguchi1024, 2023-02-02:


alexreinking, 2023-02-02:
I emailed Leo about the example we discussed in our meeting today:

```
for i in range(0, M / 4):
  for j in range(0, 4):
    for k in range(0, (4 * i + j) / 4):
      ...
```

Here, we can statically determine that `k` is in `[0, i)`, independent of `j`.

alexreinking, 2023-02-02:
Here's a test script I wrote to play around with this:

```
from z3 import *

M = Int("M")
i = Int("i")  # in [0, M/4)
j = Int("j")  # in [0, 4)
k = Int("k")  # in [0, (4 * i + j) / 4)

iter_space = And(0 <= i, i < M / 4, 0 <= j, j < 4, 0 <= k, k < (4 * i + j) / 4)

solve(iter_space)
solve(And(iter_space, k >= i))

print(simplify(iter_space))
```
It outputs:
```
[k = 0, M = 8, i = 1, j = 0]
no solution
And(i >= 0,
    Not(M/4 <= i),
    j >= 0,
    Not(4 <= j),
    k >= 0,
    Not((4*i + j)/4 <= k))
```

This shows that z3 can _prove_ that `k < i`, but its `simplify` tactic doesn't discover that on its own.

rachitnigam, 2023-09-20:
Related to the broader simplifier rewrite/capabilities question.