exo-lang/exo #345: Issue running the example

j-c-w, 2023-02-26:
Hi,

I've followed the installation instructions, and running 

`python x86_matmul.py` gives me the following error:

```
... (various correct looking output) ...
            for jo in seq(0, 2):                                
                for ji in seq(0, 8):                                       
                    C_reg[jo, ji] = C[i, 8 * jo + ji]                      
                    C_reg[jo, ji] += A[i, k] * B[k, 8 * jo + ji]           
                    C[i, 8 * jo + ji] = C_reg[jo, ji]                      
def rank_k_reduce_6x16_scheduled(K: size, C: f32[6, 16] @ DRAM,            
                                 A: f32[6, K] @ DRAM, B: f32[K, 16] @ DRAM)
:                                                                          
    for k in seq(0, K):                                                    
        for i in seq(0, 6):                                                
            C_reg: f32[2, 8] @ AVX2                                        
            for jo in seq(0, 2):                                           
                for ji in seq(0, 8):                                       
                    C_reg[jo, ji] = C[i, 8 * jo + ji]                      
            for jo in seq(0, 2):                                           
                for ji in seq(0, 8):                                       
                    C_reg[jo, ji] += A[i, k] * B[k, 8 * jo + ji]           
                    C[i, 8 * jo + ji] = C_reg[jo, ji]                      
Traceback (most recent call last):                                         
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/examples/x86_mat
mul.py", line 59, in <module>                                              
    avx = autofission(avx, avx.find('C_reg[_] += _ #0').after(), n_lifts=3)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.
9/site-packages/exo/API_scheduling.py", line 86, in __call__               
    return self.func(*bound_args.args, **bound_args.kwargs)                
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.
9/site-packages/exo/API_scheduling.py", line 1710, in autofission          
    return scheduling.DoFissionLoops(proc_c, stmt, n_lifts).result()       
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.
9/site-packages/exo/LoopIR_scheduling.py", line 2477, in __init__          
    self.proc = InferEffects(self.proc).result()                           
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.
9/site-packages/exo/effectcheck.py", line 91, in __init__                  
    body, eff = self.map_stmts(self.orig_proc.body)                        
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 135, in map_stmts
    new_s = self.map_s(s)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 203, in map_s
    body, body_effect = self.map_stmts(stmt.body)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 135, in map_stmts
    new_s = self.map_s(s)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 203, in map_s
    body, body_effect = self.map_stmts(stmt.body)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 135, in map_stmts
    new_s = self.map_s(s)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 203, in map_s
    body, body_effect = self.map_stmts(stmt.body)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 135, in map_stmts
    new_s = self.map_s(s)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 203, in map_s
    body, body_effect = self.map_stmts(stmt.body)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 135, in map_stmts
    new_s = self.map_s(s)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 148, in map_s
    rhs_eff = self.eff_e(stmt.rhs)
  File "/home/jackson/LocalProjects/PhD/Autoscheduling/exo/env/lib/python3.9/site-packages/exo/effectcheck.py", line 269, in eff_e
    buf_typ = self._types[e.name]
KeyError: C_reg_124
```

How do I debug this? Thanks,

Jackson

SamirDroubi, 2023-02-27:
Did you install it using pip (`pip install exo-lang`)? Could you try to build it from source instead using the instructions [https://github.com/exo-lang/exo#build-exo-from-source](here)? Make sure to uninstall the version you already have using `python -m pip uninstall exo-lang` before building from source.

j-c-w, 2023-02-28:
Hi Samir,

I installed using the following:

```
python -m build
pip install dist/*.whl --force-reinstall
```

Jackson

j-c-w, 2023-03-02:
Hi,

Ive' done some debugging, and it's probably because I've deleted this statement:

```
avx = stage_assn(avx, 'C[_] += _', 'C_reg')                          

```

I think I deleted it because it wasn't found --- is there a replacement for this call?

SamirDroubi, 2023-03-02:
It seems that you have an Exo build from a newer commit than the commit you are viewing the example file at. As you can see here: https://github.com/exo-lang/exo/blob/master/examples/x86_matmul.py
The call to `stage_assn` doesn't exist anymore. Can you check which commit you are using when running the example file? Can you try to pull the latest version of Exo into your local repo and try to run the example again?

j-c-w, 2023-03-03:
Sorry -- yes have confused myself while trying to get it to work --- seems to work now.  Thanks!

closed 2023-03-03 by j-c-w