exo-lang/exo #364: How to get the GEMM results use the repo?

ybai62868, 2023-03-16:
Hi, 
after reading the paper about exco, I think it is a very excellent work.
But I want to know how to get the results in the fig4 (a), about the matmul utilization for gemmini and exco.
Can you give me some hints about using the gemmini-docker embedding in this project to reproduce the results?
Thanks,

yamaguchi1024, 2023-03-21:
Hi @ybai62868 

Thanks for reading the paper, and sorry for the late reply!

To reproduce Gemmini results, you'll need access to [FireSim](https://github.com/firesim/firesim), an FPGA simulator that gives us cycle-accurate simulation for Gemmini. The most time-consuming part will be to set up FireSim and compile [chipyard](https://chipyard.readthedocs.io/en/stable/) for Gemmini.

Generating the schedule from Exo is easy. If you run pytest in the repo, Gemmini schedules are already part of the test.

Please tag me if you have further questions!

ybai62868, 2023-03-22:
Thanks for your reply!
Of course, i build the chipyard with gemmini and I use the 'spike'
simulator in gemmini, and it seems the results  (cycle) is not very correct!
Thanks a lot, i will use the FireSim to have a try!

yamaguchi1024, 2023-03-22:
@ybai62868 
spike is not a cycle-accurate simulator, so if you want to get an accurate cycle count, you'll want to use FireSim. Let us know if we can help further!

ybai62868, 2023-03-22:
Hi Yuka,

Thanks a lot for your quick response!
I think exco-lang is a so great research project and I am also interested
in designing productive programming language for hardware accelerators.
Thanks for your help. I will use FireSim to get more accurate results!
If I have any questions, I will reach out to you! ^,^

Thanks,
Yang

yamaguchi1024, 2023-03-22:
@ybai62868 Great! I'll close this issue. If you have any further questions, please email me (ikarashi@mit.edu) directly.

closed 2023-03-22 by yamaguchi1024