exo-lang/exo #448: Issue with Example

pipplo, 2023-06-09:
I'm trying to run the basic example, and when I uncomment the first block I get a compilation error. any suggestions or possible causes here?

```
exocc -o . --stem avx2_matmul x86_matmul.py
Traceback (most recent call last):
  File "/home/pipplo/miniconda3/envs/ex3.9/bin/exocc", line 8, in <module>
    sys.exit(main())
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/main.py", line 46, in main
    exo.compile_procs(library, outdir, f"{args.stem}.c", f"{args.stem}.h")
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/API.py", line 151, in compile_procs
    c_data, h_data = compile_procs_to_strings(proc_list, h_file)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/API.py", line 159, in compile_procs_to_strings
    return run_compile([p._loopir_proc for p in proc_list], h_file_name)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 305, in run_compile
    fwd_decls, body = compile_to_strings(lib_name, proc_list)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 390, in compile_to_strings
    comp = Compiler(p, ctxt_name, is_public_decl=is_public_decl)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 566, in __init__
    self.comp_stmts(self.proc.body)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 624, in comp_stmts
    self.comp_s(b)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 860, in comp_s
    self.comp_stmts(s.body)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 624, in comp_stmts
    self.comp_s(b)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 860, in comp_s
    self.comp_stmts(s.body)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 624, in comp_stmts
    self.comp_s(b)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 860, in comp_s
    self.comp_stmts(s.body)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 624, in comp_stmts
    self.comp_s(b)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_compiler.py", line 870, in comp_s
    line = mem.alloc(name, ctype, self.shape_strs(s.type.shape()), s.srcinfo)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/libs/memories.py", line 175, in alloc
    raise MemGenError(f"{srcinfo}: AVX2 vectors are not scalar values")
exo.memory.MemGenError: /home/pipplo/exo/examples/x86_matmul.py:26:16: AVX2 vectors are not scalar values
make: *** [Makefile:6: avx2_matmul.c] Error 1
```

pipplo, 2023-06-09:
I should note that I'm using the latest commit and built exo locally.

yamaguchi1024, 2023-06-09:
Hi @pipplo , thanks for raising an issue! We haven't updated the examples recently so something might be outdated. I will take a look later today!

pipplo, 2023-06-20:
Yes I was on the latest mainline at the time (not pip installed). Let me double check again

pipplo, 2023-06-20:
Ok, looks like some things have changed. Looks like you modified the example [here](https://github.com/exo-lang/exo/commit/022300b543de8c15b40bdbba2b8bf981b84716b2)

I've unblocked all the blocks. When I unblock #6 I get an error still (but it's a new error below)

```
(ex3.9) pipplo@PF40TWNV-LT:~/exo/examples$ make
exocc -o . --stem avx2_matmul x86_matmul.py
Traceback (most recent call last):
  File "/home/pipplo/miniconda3/envs/ex3.9/bin/exocc", line 8, in <module>
    sys.exit(main())
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/main.py", line 40, in main
    library = [
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/main.py", line 43, in <listcomp>
    for proc in get_procs_from_module(load_user_code(mod))
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/main.py", line 92, in load_user_code
    loader.exec_module(user_module)
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pipplo/exo/examples/x86_matmul.py", line 90, in <module>
    avx = replace(avx, 'for ji in _:_ #0', mm256_loadu_ps)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/API_scheduling.py", line 100, in __call__
    return self.func(*bound_args.args, **bound_args.kwargs)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/API_scheduling.py", line 945, in replace
    ir, fwd = DoReplace(subproc._loopir_proc, block_cursor._impl)
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_unification.py", line 83, in DoReplace
    new_args = Unification(temp_subproc, stmts, live_vars).result()
  File "/home/pipplo/miniconda3/envs/ex3.9/lib/python3.9/site-packages/exo/LoopIR_unification.py", line 681, in __init__
    raise UnificationError(f"Unification of various index expressions failed")
exo.LoopIR_unification.UnificationError: Unification of various index expressions failed
make: *** [Makefile:6: avx2_matmul.c] Error 1
```

yamaguchi1024, 2023-06-20:
@pipplo Thanks for catching this! I could reproduce the error you ran into and fixed it in the master branch.

pipplo, 2023-06-20:
Sweet, works now. Question: Do you have any tips / info on how to do debugging?

yamaguchi1024, 2023-06-20:
I usually print the code line by line and see where it doesn't behave as expected. For this example, `replace_all` was not replacing the broadcast instruction, so I thought there's something wrong with `replace_all` usage. We changed replace_all to be memory aware few months ago, and this example code was not updated after the change.

Let me know if you are interested in Exo development, always happy to chat online!

pipplo, 2023-06-21:
Sure what's the best way to connect? gmail?

yamaguchi1024, 2023-06-21:
Closing as this issue is solved

closed 2023-06-21 by yamaguchi1024