exo-lang/exo #475: Halide Example Notebook

rachitnigam, 2023-07-21:
I'm working through the `blur` example from the Halide PLDI paper and attempting recreate the schedules from the paper. First crack at writing a `proc`-based implementation:
```py
@proc
def blurx(inp: f32[32, 32] @ DRAM, out: f32[32, 32] @ DRAM):
    for y in seq(0, 32):
        for x in seq(0, 32):
            left: R;
            right: R;
            if x == 0:
                left = 0.0
            else:
                left = inp[y, x - 1]

            if x == 31:
                right = 0.0
            else:
                right = inp[y, x + 1]
            out[y, x] = (left + inp[y, x] + right) / 3.0

@proc
def blury(inp: f32[32, 32] @ DRAM, out: f32[32, 32] @ DRAM):
    for y in seq(0, 32):
        for x in seq(0, 32):
            top: R;
            bottom: R;
            if y == 0:
                top = 0.0
            else:
                top = inp[y - 1, x]

            if y == 31:
                bottom = 0.0
            else:
                bottom = inp[y + 1, x]
            out[y, x] = (top + inp[y, x] + bottom) / 3.0

@proc
def prog(inp: f32[32, 32] @ DRAM, out: f32[32, 32] @ DRAM):
    blurx_out: f32[32, 32] @ DRAM
    blurx(inp, blurx_out)
    blury(blurx_out, out)
```

An immediate problem here is that performing something like a `compute_at` requires either: (1) changing the signature of the `blurx` method or inlining it so that we can resize the input `inp` to the proc. If we end up doing the latter, then the `proc` abstraction seems a bit useless.

jrk, 2023-07-21:
I think you're hitting a classic confusion here, conflating boundary conditions *on the edge of the image* (pad with 0 or whatever) with boundary handling *at the edge of tiles* (you need to enlarge the input tile, or perhaps read from a neighboring tile).

Try making this have an outer loop over many 32x32 output tiles, but interleave the two stages per-tile.

gilbo, 2023-07-22:
Apart from that, it also makes sense that you have to inline function calls to do Halide-style optimizations. The basic imperative analog of a Halide func is a memory allocation and loop nest, not a function call. The proc boundary might align (operationally) with a Halide pipeline boundary.That said it’s still useful as a way to organize the original program in Exo.Additionally, Alex was interested quite a while ago in having a functional language that used non-standard control flow to model the Halide computation orders while maintaining some kind of truly functional boundary between funcs/stages

jrk, 2023-07-23:
@gilbo For context, we discussed trying to use `proc`s as much as possible to keep things organized, rather than just trying to ram through a lowered loop nest like others had tried before, so this was intentional.