exo-lang/exo #493: Problems handling tail loops

SamirDroubi, 2023-09-23:
A common tail loop that shows up during scheduling is something like:
```
for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg[ii] += x[ii + 8 * io] * y[ii + 8 * io]
```
With some scheduling you can get to:
```
        reg_2: f32[8] @ AVX2
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg_2[ii] = x[ii + 8 * io]
        reg_3: f32[8] @ AVX2
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg_3[ii] = y[ii + 8 * io]
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg[ii] += reg_2[ii] * reg_3[ii]
```
However, this is not exactly what you want because there are no native instructions that are exactly semantically equivelant to these loops:
1. Mask operations in AVX2 and AVX512 really write either zero or values provided by a mask, not garbage.
2. Mask operations in AVX2 only exists for loads/stores/gather/scatter i.e. memory operations.
We can implement Exo instructions that are semantically equivelant to these loop by going down to C which is less ideal. Properly handling this requires reasoning about unwritten values by the original algorithm.

Ideally, I would want to go to:
```
        reg_2: f32[8] @ AVX2
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg_2[ii] = x[ii + 8 * io]
            else:
                reg_2[ii] = 0.0
        reg_3: f32[8] @ AVX2
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg_3[ii] = y[ii + 8 * io]
           else:
               reg_3[i] = 0.0
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg[ii] += reg_2[ii] * reg_3[ii]
```
Then, reason that if condition in the fma loop is not neccesary now to since we would be accumulating zeros not garbage values:
```
        reg_2: f32[8] @ AVX2
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg_2[ii] = x[ii + 8 * io]
            else:
                reg_2[ii] = 0.0
        reg_3: f32[8] @ AVX2
        for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg_3[ii] = y[ii + 8 * io]
           else:
               reg_3[i] = 0.0
        for ii in seq(0, 8):
              reg[ii] += reg_2[ii] * reg_3[ii]
```

Another limitation even if the above is fixed is that you would still need to handle to implement all control-related logic within the C instruction implementation which is 
1. Puts more things into unsafe C code
2. Results in duplicated code (the same mask vector instructions needs to be generated for the X and Y load)
This second issue was pointed out by #91 as well.

What I would expect doing:
```
for ii in seq(0, 8):
            if ii + 8 * io < n:
                reg[ii] += x[ii + 8 * io] * y[ii + 8 * io]
```
Rewrite the conditional (which we also don't support):
```
for ii in seq(0, 8):
            if ii < n - 8 * io:
                reg[ii] += x[ii + 8 * io] * y[ii + 8 * io]
```
Bind the conditional lhs and rhs:
```
lhs_cond: index[8] @AVX2
for ii in seq(0, 8):
      lhs_cond[ii] = ii
rhs_cond: index[8] @AVX2
for ii in seq(0, 8):
      rhs_cond[ii] = n - 8 * io
lt: bool[8] @AVX2
for ii in seq(0, 8):
      lt[ii] = lhs[ii] < rhs[ii]
for ii in seq(0, 8):
            if lt[ii]:
                reg[ii] += x[ii + 8 * io] * y[ii + 8 * io]
```
Now, some as above to go from
```
for ii in seq(0, 8):
            if lt[i]:
                reg[ii] += x[ii + 8 * io] * y[ii + 8 * io]
```
to
```
reg_1: f32[8] @ AVX2
for ii in seq(0, 8):
            if lt[ii]:
                reg_1[ii] = x[ii + 8 * io]
            else:
                reg_1[ii] = 0.0
reg_2: f32[8] @ AVX2
for ii in seq(0, 8):
            if lt[ii]:
                reg_2[ii] = y[ii + 8 * io]
            else:
                reg_2[ii] = 0.0
for ii in seq(0, 8):
           reg[ii] += reg_1[ii] * reg_2[ii]
```

Now, all my loops map to native AVX2 instructions!

yamaguchi1024, 2023-09-24:
The unification issue seems to be coming from this code, because we are unifying LHS and RHS separately:
https://github.com/exo-lang/exo/blob/master/src/exo/LoopIR_unification.py#L1128

SamirDroubi, 2023-09-25:
For context, the unification problem @yamaguchi1024 mentioned is this issue: #496 which showed up with tail loops