exo-lang/exo #517: Srcinfo and error reporting

SamirDroubi, 2023-10-07:
Currently, we maintain srcinfo information throughout scheduling. Whenever, we report errors, we do so with respect to srcinfo.

From my experience in scheduling, this is not very valuable information. It doesn't help me with debugging at all; especially if it is an error at codegen.

Since we are in a rewrite based system, it makes more sense to reference the current state of the program (or the diff if it is a scheduling operation) and where exactly did the error occure. This should especially be easier to implement now with cursors; there is convenient tooling to point at places in the code where errors occured.

Moreover, I don't think we maintain srcinfo within the compiler in any consistent ways. Maintaining proper srcinfo also requires coming up with some sort of policies in various places. e.g. what happens when you fuse two loops. I would also imagine that if we plan to maintain srcinfo in a meaningful we would have to design their policies to be consistent with cursors which presents more complexity desiging any scheduling operation.

Finally, I would imagine deprecating srcinfo would reduce the working set of exo procs by a good amount since we maintain that information for every single node. I don't have a good esitmate on that though or the impact of such reduction on compilation time. We should measure this though if we ever choose to deprecate it.

rachitnigam, 2023-10-07:
Thanks for writing this up! In general, this is a tricky problem that all compiler (not just rewrite-based ones) have to deal with. Even in a standard compiler, the intermediate representation is derived from normalization of surface constructs which requires the same kind of tracking. There are different trade-offs you can make w.r.t. to the amount of information tracked and the precision of the error messages.

For example, the Rust compiler uses the [`Diagnostics`](https://rustc-dev-guide.rust-lang.org/diagnostics/diagnostic-structs.html) struct for reporting error messages. The error messages, especially when generated from a type error, often represent a tree of rewrites from source-level AST to the internal representation which allows Rust to provide error messages that meaningful for a particular context.

As a maximalist, strawperson proposal, maybe the thing to do is tracking *all* of the source information generated when rewriting programs. For example, when a loop is fused, we add a `srcinfo` node that literally says that this node is derived from these other two loops.

SamirDroubi, 2023-10-07:
I think the difference here is that I the user was the person who drove the rewrites to this point vs in rust the compiler did ao, so I don't particularly care about the original procedure. I care about where I am at now and where the problem is here.

rachitnigam, 2023-10-07:
Hm, but that assumes that you, as the user, have seen each rewrite. This is not going to be true when using the abstracted scheduling operators. For example, if there is an error in the middle of `vectorize`, which version of the program should the error be reported on. Ideally we'd want the diagnostics mechanism to have a hook that says "start tracking all the rewrites from now on" and report the transformations that occurred within `vectorize`, showing the trace of rewritten programs which might give an insight into what caused the error.

A similar problem might arise when building Halide-like automation which parameterizes the lowering process and triggers bound inference on some IR which is not meaningful to someone working with a surface program.

SamirDroubi, 2023-10-07:
Now we are discussing a different problem. What you are describing is the next problem to solve if we implement what I described above. The confusion I think is that srcinfo in Exo is information about line and column numbers in the Python source file from which the current AST node originated.

SamirDroubi, 2023-10-07:
What I am generally saying is that if the user asks Exo to do some action `A` (scheduling, codegen, other?) on some procedure `P`, then error reporting should be with respect to procedure `P` which is not always the case. The extension of this is what you described: I ask an abstraction to do something on a program `P` and error reporting should be on `P`.