exo-lang/exo #547: GPU support in Exo

skeqiqevian, 2023-12-29:
Wanted to preface by saying that this design is definitely not done. However, I wanted to describe the proposal based on discussions with William and get some feedback. 
 
# ExoIR representation of CUDA abstractions
CUDA has three primary abstractions that we want to support: memories, parallel hierarchy, and synchronization. We first describe how we represent these in ExoIR. Later, we will describe the necessary safety checks to prevent users from generating bad CUDA code. Ideally, we want to prevent both data races and deadlocks.

## Parallel Hierarchy
We will represent parallel block/thread loops as parallel loops with special annotations (e.g. `@THREADS` or `@BLOCK`). In CUDA, these loops are always implicit because users are tasked with writing thread programs. In our programming model, we require users to explicitly write parallel loops. Users may write consecutive parallel loops, e.g. the following correspond to running threads `0-7` all in parallel:
```python
for i in par(0, 4): @THREADS
    ...
for i in par(4, 8): @THREADS
    ...
```

Code generation from this programming model to the CUDA programming model is simple. Each such block/thread loop actually corresponds to an `if` statement which predicates over the specified loop range:

<table>
<tr>
<th>CUDA C++</th>
<th>Exo</th>
</tr>
<tr>
<td style="vertical-align:top">

```python
for i in par(0, 5) @ BLOCKS:
    for i in par(0, 5) @ THREADS:
        ...
```
</td>
<td>

```cpp
if (blockIdx.x < 5) {
    if (threadIdx.x < 5) {
        ...
    }
}
```
</td>
</tr>
</table>

### Excessive generation of if statements
This approach may generate vacuously true `if` statements (e.g. when iterating over all threads), so we should prune those.
In particular, unless there is complex block-level synchronization, all of the block-level loops will likely generate vacuously true `if` statements.

## Memory
We will define specialized memory classes for the shared memory (`SHARED_MEMORY`) and thread-local registers (`THREAD_REGISTER`), just as we did for AVX vector registers.

These memories we require additional checks:
 - shared memory should only be accessible by threads within the same block.
 - thread-local registers should only be accessible by that thread

## Synchronization
We want to give users control over synchronization.
Thus, it is the user's responsibility to properly insert synchronization primitives into their code.
At compilation time, we will verify that the user inserted syncs properly before generating CUDA code.

In CUDA code, we can perform synchronization over arbitrary predicates (like below).
However in Exo, we will need to restrict ourselves to predicates of index expressions.

As a design choice to avoid reasoning about complicated synchronization patterns, we choose to make synchronizations happen *outside* of the parallel for loops. Thus, Exo code will
<table>
<tr>
<th>CUDA C++</th>
<th>Exo</th>
</tr>
<tr>
<td>
[... 196 lines cut ...]
}
```
</td>
<td style="vertical-align:top">

```python
bar: CUDA_BARRIER @ SHARED_MEMORY
for i in par(0, 1): @ THREADS
    init(&bar, ...)
sync(0, N)

for i in par(0, N): @ THREADS
    # code before arrive
sync(0, N)

for i in par(0, N): @ THREADS
    compute(...)
sync(0, N)

for i in par(0, N): @threads
    # code after wait
```
</td>
</tr>
</table>

gilbo, 2024-01-22:
Hi Kevin,

Here are the two big comments/questions:
1. Is the plan that this will be developed in a fork from Exo for the purposes of a Master's project or is this a proposal for a core Exo feature?  Maybe a fork would give you more freedom to quickly start experimenting, and not get you stuck in trying to have a consistent design.  For example, you could just create an "Exo-CUDA" language instead of worrying about how to externalize the CUDA programming model.
2. My original understanding of this project was that the idea was to explore very conservative (w.r.t. concurrency/synchronization) GPU support for Exo.  The sketch here seems to be headed towards supporting synchronization primitives, which will be non-trivial because it could invalidate the soundness of all currently existing analyses/program transforms.  If it doesn't, it's unclear to me exactly why it doesn't.

You may want to look at something called the Bulk Synchronous Parallel (BSP) model as a potential grounding for what you are proposing.

Here are some further comments jotted down while reading:
* Your CUDA C++ and Exo labels on the first table are swapped
* Why would we define a `THREAD_REGISTER` memory type; don't you merely want at `GPU_DRAM` memory?
* What are the semantics of barriers, and how many other scheduling primitives are there?  Can you do a version of this that doesn't use synchronization?
* "Our proposed programming model doesn't require an entire thread program to be in a single loop over threads" This is inconsistent with the CUDA programming model, right?  So then, are you proposing to somehow transform the whole Exo program automatically to be consistent with this idea?  If so, why is that an automatic transform instead of a scheduling transform?
* "(they may still read from shared read-only memory)" -- note that shared memory is not persistent across CUDA kernels.  Consequently, you will never be allowed to read anything meaningful from shared memory.
* "For each synchronization, we union all of the access lists of those threads, and assign each thread that list." --- why is this the right thing to do?
* re: analysis example; how does this work when the loop bound is a variable rather than a constant?
* why does the barrier have a memory location in the last example?

yamaguchi1024, 2024-01-29:
- It will be helpful to mention the implicit synchronization proposal and talk about its pros and cons (the pros will be that the analysis is simple, and the cons will be the performance compromise due to the inflexibility).
- I think you want to talk about adding loop annotations (`@threads`, `@openmp`, and so on) and how that will affect the backend code generation
- I guess we want a "procedure annotation for rewrite no-ops" for explicit synchronization call (`sync(..)`), where users can define their own backend checks externally. It will also be useful for handling prefetch since it's also a no-op.
- The analysis will be more complex if threads and blocks can be non-integer-constant.

yamaguchi1024, 2024-01-29:
Notes from a discussion with Kevin and William:
- Even though consumer and server GPUs have slightly different design points, we probably want to focus on server GPUs for now
- Synchronization predicates could be arbitrarily complex, but expressing just a range might be sufficient for many workloads
- We should keep in mind that there are warp synchronous operations and non warp synchronous operations