exo-lang/exo #778: Does annotation @instr support custom instruction?

TJU-PanYizhe, 2025-04-24:
RISC-V allows developers to add new custom instruction for specific applications.

Does annotation @instr in EXO compiler support this kind of custom instruction utilization?

For example, `vmadot` is introduced on my RISC-V CPU hardware for matrix multiplication. 

If the compiler haven't supported the custom instructions(compiler cannot identify `vmadot` or generate `vmadot`), could I use `insn` with asm when using annotation @instr

yamaguchi1024, 2025-04-24:
Yes! Indeed, letting users define custom accelerator instructions was the whole point of `@instr` when we designed it.

TJU-PanYizhe, 2025-04-27:
> Yes! Indeed, letting users define custom accelerator instructions was the whole point of `@instr` when we designed it.

@yamaguchi1024 

Thanks for your reply! `vmadot` instruction introduces some constraint of RISC-V register utilization, is there any method or annoation to define this kind of constraint?

Or do I have to write assembly code by my self?

TJU-PanYizhe, 2025-04-29:
> Btw, did you take a look at the RVM tutorial ( https://exo-lang.dev/tutorial.html ) and its code ( https://github.com/exo-lang/exo/tree/main/examples/rvm_conv1d )?

@yamaguchi1024 Yes, I've read the tutorial and its code.

`vmadot` 's format is `vmadot rd, rs1, rs2`, and it requires the number of rd must be even(such as v0, v2, v4, v6, ..., rd%2==0).

Additionally, `vmadot` store the result in rd and rd+1 two registers automatically without any other instructions.

How to express `vmadot` with @instr please?

yamaguchi1024, 2025-04-29:
I see. Do you have a C intrinsic for `vmadot` or do you only have an assembly?

Is `vmadot` computing a dot product? If so, I think it'll look sth like
```
@instr("<C or assembly string for vmadot >")
def vmadot(dst: f32, rs1: [f32][8], rs2: [f32][8]):
  for i in seq(0, 8):
    dst += rs1[i] * rs2[i]
```

TJU-PanYizhe, 2025-04-30:
> I see. Do you have a C intrinsic for `vmadot` or do you only have an assembly?

@yamaguchi1024 There is only assembly for `vmadot`, like `vmadot v4, v0, v2`, which calculates the multiplication of two matrices stored in vector register `v0` and `v2`, and store the result in `v4` and `v5`

`v5` is not explicitly assigned in the code(`vmadot v4, v0, v2`), which explains why the number of `rd` must be an even number.
Meanwhile, `rd` should not be equal to `rs1` or `rs2`

I can use `vmadot` with the C++ code below:

```C++
size_t vlenb = __riscv_vlenb();
size_t vlen = vlenb * 8;
const size_t vlmax_e8m1 = __riscv_vsetvlmax_e8m1();
int8_t a[vlmax_e8m1];
int8_t b[vlmax_e8m1];
for (int i = 0; i < vlmax_e8m1; i++)
{
    a[i] = 1 + i;
    b[i] = 1 - i;
}
const size_t vlmax_e32m1 = __riscv_vsetvlmax_e32m1();
int32_t c[vlmax_e32m1];
int32_t d[vlmax_e32m1];
asm volatile(
            "vsetvli x7, x0, e8, m1, tu, mu\n\t"
            "vle8.v v0, (%[a])\n\t"
            "vle8.v v2, (%[b])\n\t"
            "vmadot v4, v0, v2\n\t"
            "vsetvli x7, x0, e32, m1, tu, mu\n\t"
            "vse32.v v4, %[c]\n\t"
            "vse32.v v5, %[d]\n\t"
            : [c] "=g"(c), [d] "=g"(d),
            : [a] "r"(a), [b] "r"(b),
            : );
```

The pseudo code of `vmadot` is as follows:

```C++
void native_vmadot(int8_t *a, int8_t *b, int32_t *c, int32_t *d, int i8s)
{
    for (int i = 0; i < i8s / 4; i++)
    {
        c[i] = -1;
        d[i] = -1;
    }
    for (int i = 0; i < 4; i++)
    {
        for (int j = 0; j < 4; j++)
        {
            // cout << "i = " << i << ", j = " << j << endl;
            if (i <= 1)
            {
                for (int k = 0; k < 8; k++)
                {
                    c[i * 4 + j] += a[i * 8 + k] * b[j * 8 + k];
                }
            }
            else if (i > 1)
            {
                for (int k = 0; k < 8; k++)
                {
                    d[(i - 2) * 4 + j] += a[i * 8 + k] * b[j * 8 + k];
                }
            }
        }
    }
}
```

yamaguchi1024, 2025-04-30:
I see, got it!
In your code, `v4` is `c` and `v5` is `d`, right?
- Since your interface is inline assembly, handling the register allocation will be necessary, which you can do in Exo (basically just Python) or in C compilers. Using C compilers probably makes more sense if you're eventually going to do that anyway.
- We are actively working on supporting variable vector length architectures (RVV and SVE VLA) in Exo, and also working on changing the Exo object code frontend and the analysis entirely. The tricky part for supporting RVV is not the individual instructions, but instead expressing and analyzing the dynamic loop bounds, which requires an analysis change.
- We will be working on supporting SME in a few months, and we're collaborating with UC Berkeley folks who are taping out the RISC-V outer-product matrix extension, which is similar to SME. In the fall, I'm hoping that we can support the outer product engine and test some code on the taped-out chip.
- `vmadot` seems like a nice extension to RISC-V, and I'm interested in supporting and testing it as well. But that has to first wait for our RVV backend to be fully working, which should happen in a month or so.

If you're interested in talking more, I would be happy to chat on zoom.

TJU-PanYizhe, 2025-05-07:
> In your code, `v4` is `c` and `v5` is `d`, right?

@yamaguchi1024 Correct!

> * Since your interface is inline assembly, handling the register allocation will be necessary, which you can do in Exo (basically just Python) or in C compilers.

Is there any example code to do that in Python? How to do it in Python?

> * Using C compilers probably makes more sense if you're eventually going to do that anyway.

If I understand right, you mean modifying the backend code of C compiler and re-compile the compiler to support the new instructions(`vmadot`)?

yamaguchi1024, 2025-05-07:
We have a (very limited) example for Intel AMX [here](https://github.com/exo-lang/exo/blob/main/src/exo/libs/memories.py#L283), but you probably want sth more sophisticated than just incrementing and decrementing the pointer. It's basically just a vanilla register allocation problem.

Yes, I think supporting a new instruction as an intrinsic is not that hard, but plugging in the register allocation seems harder.